adafruit/Adafruit_CircuitPython_SimpleIO #3: servo control

ladyada, 2016-12-08:
sets 50 hz pwm with pulse output, from 500us to 2500us

caternuson, 2016-12-08:
RC sevos suffer from lack of standard and confusing/conflicting info. 50Hz seems to be pretty universally accepted. The main source of confusion being the PWM timing range and resulting amount of servo rotation. 1000-2000us seems to be the accepted safe range with anything beyond those limits being servo dependent. Also some confusion on what 1000-2000us produces in servo output, with some info saying 90deg and other saying 180deg.

So maybe make the timing limits user configurable, but with default values of 1000-2000us.

ladyada, 2016-12-08:
could vargs it so default is us timing, and angle=90 is like, degrees but i agree i prefer passing in timing not angle

tannewt, 2017-08-09:
Done in f4d2736d307d6bcda00debf686bd9a7d7a7c43ed. Thanks @brentru

closed 2017-08-09 by tannewt