adafruit/Adafruit_CircuitPython_SimpleIO #19: tone(pin, freq, duration)

microwattbott, 2017-11-27:
While test a early beta version of  piezo_music.py,  
I struggled to understand that freq had to be an integer.  
A float value is not going to make the tone() function work.  
@tannewt  suggested that we try to "fix tone to either handle it or produce a friendly error message."

RE:   \Metro Experimenter Guide\METROX-Examples-and-Project-Sketches-master\circuitpython\CIRC06\piezo_music.py

brentru, 2017-12-23:
Thanks for the feedback! I'm going to wire up the circuit for piezo_music.py and fix it per @tannewt's suggestions (thx for including those btw!)

brentru, 2018-01-05:
Tone function has been changed, and piezo_music.py in the MetroX circuit examples has been updated.

closed 2018-01-05 by brentru