adafruit/Adafruit_CircuitPython_SimpleIO #33: Remove Servo

tannewt, 2018-08-15:
[adafruit_motor.servo](https://github.com/adafruit/Adafruit_CircuitPython_Motor/blob/master/adafruit_motor/servo.py) is better because it also supports continuous Servos.

Guides will also need to be updated.

brentru, 2018-08-15:
Guides which reference simpleio servo:

- [ ]  https://learn.adafruit.com/circuitpython-essentials/circuitpython-servo
- [ ]  https://learn.adafruit.com/combo-dial-safe-with-circuit-playground-express?view=all#code-with-circuitpython

sommersoft, 2018-10-02:
This looks to be fixed by #34. Reopen if I'm mistaken...

sommersoft, 2018-10-14:
On Discord, user Faeylayn, reported that the [CPX Perk Up Ears](https://learn.adafruit.com/perk-up-ears/code) guide still references `simpleio.servo`. That guide needs to be updated.

tagging for attention: @dastels

dastels, 2018-10-14:
I'm on it.

dhalbert, 2018-10-17:
[ ] Another guide that uses `simpleio.Servo`: https://learn.adafruit.com/trinket-gemma-servo-control/circuitpython-code
Author is @mikeysklar

mikeysklar, 2018-10-17:
I'll go ahead and update this guide to use adafruit_motor.servo in place of simpleio.Servo.

mikeysklar, 2018-10-17:
PR submitted and approved for Trinket Gemma Servo Control. The guide has been updated to use adafruit_motor.servo and an update was posted in the forum.

kattni, 2018-12-02:
Looks like this was taken care of. We can reopen it if we run into more instances needing an update.

closed 2018-12-02 by kattni