adafruit/Adafruit_CircuitPython_SimpleIO #35: Update simpleio.tone to match 3.x audioio api

kattni, 2018-10-02:
`simpleio.tone` is using a previous `audioio` API. We need to update it to check the version number, use the 3.x+ `audioio` API, and if it finds 2.x, to use the previous API. See https://github.com/adafruit/Adafruit_CircuitPython_CircuitPlayground/blob/master/adafruit_circuitplayground/express.py#L622 for an example of using version numbers to utilise multiple APIs in a single funciton.

The issue is here: https://github.com/adafruit/Adafruit_CircuitPython_SimpleIO/blob/master/simpleio.py#L57

This will need to be tested by using `simpleio.tone` to produce a tone using a non-PWM pin.

caternuson, 2018-10-02:
See #36

kattni, 2018-10-03:
Fixed, thank you!

closed 2018-10-03 by kattni